marazmista/radeon-profile #235: Something wrong with graph tab

baryluk, 2020-10-05:
Hi,

I just built the radeon-profile for the first time (I learned about it randomly from some random screenshot in ROCm issue thread).

Anyhow this is issue:

Running at 1f98740b95b1de390d981f7894b93c5b46a44518

Running as root

This is the graph tab:

[image]

[image]

[image]

That doesn't look right.

I am using Linux 5.7.6, with AMD Radeon R9 Fury X and `amdgpu.ppfeaturemask=0xffffffff`. I am vanilla amdgpu from upstream (I do have ROCm, but without rocm-dkms).

Most of the other tabs and values all look good, even fan control works.

marazmista, 2020-10-17:
I think some default plots should show but anyway, when you click on Enable left scale or Enable right scale you should be able to choose which values you want for plots

baryluk, 2020-10-20:
Indeed, it is possible to add new plots. It was a bit non intuitive. Maybe show instead all available data, always, and automatically enable left or right depending if anything is added. No need for the extra `Enable left scale`, `Enable right scale`.

[image]

By default the "Graphs" shows no graphs for me. Kernel 5.7.6, Fury X, just like before.

marazmista, 2020-11-01:
I checked with fresh configuration and indeed, there is no default plots. I pushed a commit where left scale is enabled by default when user create new plot (https://github.com/marazmista/radeon-profile/commit/0fd5b8dd326957e6fa8afda2bba5fc4c48fd0de4 ).

baryluk, 2020-11-16:
@marazmista That is nice. It is now much more obvious what is going on.

I added few graphs, and they do look to work. The graph is kind of ugly, with no legend and all lines having same colour, but that is a topic for another time.

closed 2020-11-16 by baryluk