marazmista/radeon-profile #238: Not sure if this is an issue or not

mrgfy, 2020-10-15:
Hi, First i want to say, this is an amazing tool and I use it everyday. Much respect to you all. I'm just curious as to how my radeon-profile looks different from the screenshot posted on the repo's readme page? I apologize for my ignorance if perhaps it's an effect of my gtk theme, but i don't think so.
[image]

marazmista, 2020-10-17:
Hi, thanks.
Screenshot in readme is pretty old (from 2017). To be hones, only major difference I see is buttons on the left side are different (but only for newer cards). Old buttons are still valid for some systems and will show up like on screenshot.

mrgfy, 2020-10-17:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA256

ty for clearifying that, i was just confused. again awesome tool. You all rock!

Sent with ProtonMail Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

-----BEGIN PGP SIGNATURE-----
Version: ProtonMail

wsBzBAEBCAAGBQJfix4dACEJECj+QiNMjKz8FiEEu3ynjnrAi1V/Ga+cKP5C
I0yMrPyx1Af+LV3AwJsGlRmYto6zCBXooxuG+07jsUPY7oM87VEI8rgaxWnC
xNkMFq6L9MjTH0IcCBmgGXukmDYieg0iLI8Kcic3Y0m0YtQitjpdsiA9UGW4
RcpufLni2GXYM2GrvueVfYBTPdQeiiNpjeLtIPkMu09lHzzRY9AZMyeHX5Dy
94CrK7TT7DYQ5R6PmqxXsJdC82HIqpL8MRNptLUieAUEV3EJNpTZ08rK583F
//sLEf1j6rdK9gTECxIqaeHDGdjlL9SLijCdUsQ4WXslynFWV35GTaAdnmdQ
aTWwPQZnKf8f0bcPj8HWjx2PyqjXDjkQaz7pn/OOe2PzQ799NgI5Hg==
=AAbX
-----END PGP SIGNATURE-----

closed 2020-10-31 by marazmista