marazmista/radeon-profile #242: [Suggestion] Start in background

MithicSpirit, 2020-11-28:
There should be a switch like `--background` that makes radeon-profile begin without creating a window. This would be useful for me because I want it to begin controlling my fans when I log in but it's kind of annoying that it creates a window.

void-witch, 2020-12-22:
this. i assumed this would be the real purpose of the daemon, but i guess not. using a daemon just to get around file permissions seems like a silly thing to me, this would be a lot more useful. my card's default curve is too aggressive in some places, and not enough in others. i've got a good, safe curve, so there's no reason for me to have to have a window open

mainmachine, 2020-12-22:
@void-witch Leveraging a system service or daemon in this way for the purpose of preventing direct user access to protected files, devices, etc. is a pretty well-established standard at this point, regardless of OS. Take a look at Network Manager as an example, but there are dozens of other examples you can look at in systemd alone.

void-witch, 2020-12-22:
@mainmachine what i mean is, the work required to implement persistence of settings is already done, no reason not to have it

mainmachine, 2020-12-28:
It seems like you are looking for the `Start minimized` and `Minimize to tray` options as pictured... or am I missing something here?

[image]

void-witch, 2020-12-29:
personally, i'd like to just have the daemon persist settings, so i don't need a gui application at all