marazmista/radeon-profile #247: Hainan Chip overheating

skarmiglione, 2020-12-26:
Hello there, i hope you are fine, i have a problem on Ubuntu with a discreet gpu HAINAN CHIP, because it get reboot always i try to play DRI_PRIME=1 games in an Asus x555qg:

Graphics: Device-1: Advanced Micro Devices [AMD/ATI] Wani [Radeon R5/R6/R7 Graphics] driver: amdgpu v: 5.6.19
Device-2: AMD Sun XT [Radeon HD 8670A/8670M/8690M / R5 M330 / M430 / Radeon 520 Mobile] driver: amdgpu v: 5.6.19
Display: x11 server: X.Org 1.20.8 driver: amdgpu,radeon resolution: 1366x76860Hz, 1280x102460Hz
OpenGL: renderer: AMD Radeon R7 Graphics (CARRIZO DRM 3.40.0 5.4.0-58-generic LLVM 11.0.0) v: 4.6 Mesa 20.3

i have give DC and DSP parameters to the kernel grub but can't really configure a lot of things there, i did try corectrl before radeon-profile and there i only had 2 options "Low" = 300mhz(clk) & 300mhz(mclk);  "High" 1030mzh (clk) & 900mhz(mclk) oh and "Auto" but this is always the high state 1030mhz x 900mhz same as High <- this is what burn my laptop btw. Until here i just decide to unistall corecontrol and try with Radeon Profile, and it give me more information and a new mode more than core ctrl is working here 400 mhz (clk) & 600(mclk) (that is making playable some games and i really apreciate that from radeon profile thanks) But, i always have the doubt cause if this gpu can get to 1030mhz with a 105°c of temperature, maybe there is a way to  get less than High state 1030mhz x 900mhz and maybe more than Auto - Battery state with 400mhz x 600mhz , i suppose it can be stable on 600mhz x 600mhz maybe? and i will happy with that, but doing $sudo echo <parameters from internet> > /pp_dsp_* files did'nt work for me never yet :( 

Now the other thing i was thinking are a way to put ranges on the events but the less temperature event is working over the high temperature event is not working as i think with profile "high" on less than 90°c and a way to up the cycles number to get less than 90°c.... Thanks for atention, sorry for my bad english.

skarmiglione, 2020-12-26:
https://github.com/marazmista/radeon-profile/issues/164

skarmiglione, 2020-12-28:
https://gitlab.freedesktop.org/drm/amd/-/issues/1419

JoneKone, 2021-03-05:
You should replace your thermal compounds in your GPU.

skarmiglione, 2021-03-26:
https://gitlab.freedesktop.org/drm/amd/-/issues/1419#