marazmista/radeon-profile #249: Radeon-profile seems to "loose" fan-control sometimes

typoworx-de, 2021-01-04:
I'm new to radeon-profile and it's a great tool. Thanks for it!

But fiddling around with fan-control the software sometimes looses ability to control fan and changing profile-stuff does work but there's no more reaction (f.e. fan does not start to use "fixed 20%"). Restarting the radeon-profile-daemon nor radeon-profile does not work. Only restarting the whole machine seems to restore functionality.

I think there may be something broken with fan device-control and manual mode?

This is the GPU I'm using this tool with:
[AMD/ATI] Tobago PRO [Radeon R7 360 / R9 360 OEM]
```
lspci -v | grep VGA
01:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Tobago PRO [Radeon R7 360 / R9 360 OEM] (rev 81) (prog-if 00 [VGA controller])
```

typoworx-de, 2021-01-04:
Tested again. Even only starting the tool running a previously configured fan-profile will stop working after about ~15 minutes after applying. Only reboot brings back functionality.

dmesg is flooded with this message in intervals:
```
[ 1440.283549] manual fan speed control should be enabled first
```

niarbeht, 2021-01-25:
Could it be related to https://gitlab.freedesktop.org/drm/amd/-/issues/1408 ?

PorcelainMouse, 2023-04-12:
I'm getting this now.