marazmista/radeon-profile #255: Support for multi-GPU control?

eamondo2, 2021-02-07:
My config is a little less than standard, so I understand if this is not something that is typically supported. I've got a 5700XT (AMD Reference model) and a Vega64 (Also AMD Reference model) running in the same system. 
Both cards show up in the radeon-profile app, listed as GPU 1 and GPU 0 respectively. (Which is a little odd, because the 5700XT is in the primary PCIe slot, and is controlling the monitors, while the Vega64 is in the secondary PCIe slot, and has no monitors attached). However, unless I have the GPU selected inside of radeon-profile, the fan settings/power settings do not apply. I.E. if I select GPU 0 and set the fan curve higher, configure it to run in "high" settings, and then swap to GPU 1, GPU 0 drops clock speeds and the fans drop back to the auto settings. GPU 1 then gets the OC/performance/fan settings applied.
I was thinking that perhaps I need an instance of radeon-profile-daemon running per GPU? Or perhaps there's some other fundamental config. issue I'm missing here.
I am running the latest versions of both the daemon and the application from the git repo, though this behavior is also present with the official version from the Ubuntu ppa.

eamondo2, 2021-02-11:
As a small update/addendum to the above, I was able to recompile both radeon-profile and radeon-profile-daemon and force them to use a slightly different ``serverSocketPath`` address such that I could run two instances of both the client and the daemon, and while there are probably numerous reasons that's a bad idea, it seems to have worked reasonably well as a proof of concept.
Running two separate daemon processes as well as clients allows me to have proper fan/clock control for both cards at the same time, though the implementation is a bit hacky. Currently I'm just manually changing the hardcoded string and launching the build result directly. I'm going to look into patching in a method for both radeon-profile-daemon and radeon-profile to allow loading the `serverSocketPath` from either an env var or some other external config source, and perhaps restricting each instance to a specific GPU such that the clients/daemons don't attempt to overwrite each other's settings on accident.

Oxalin, 2023-01-27:
Radeon profile doesn't support "managing" multi-GPU. It manages one GPU at a time for now. I'm working on forks where the config settings will be saved and restored per GPU and I think I'll make some test to see if we could run two (or more) "GPU profiles/configs" in parallel. I may have to rework the daemon too though.

As you stated, the daemon only creates one socket. It should be enough to communicate between RP and RPd for as many GPU as needed with some tweaks. First, RP must manage a config for each GPU instead of a single one for all the GPUs (right now, it flushes the UI each time we change the GPU list and changes all the settings). Then, the daemon must understand that it is managing more than one GPU at a time (which I think it can probably already do).

I'm already working toward saving and restoring the settings per GPU. I have the intention of working to support multi-GPU.