marazmista/radeon-profile #256: Documentation: Packages to install on OpenSuse Tumbleweed 2021-02

7eggert, 2021-02-08:
(because one of them is not that obvious to search for if you don't know the name)

I needed to install:
patterns-kde-devel_qt5
libQt5Charts5-devel-5.15.2-1.2.x86_64

I did have a lot of development packages installed, so the list may be incomplete for building.

Shadowolf7, 2021-11-29:
> (because one of them is not that obvious to search for if you don't know the name)
> 
> I needed to install: patterns-kde-devel_qt5 libQt5Charts5-devel-5.15.2-1.2.x86_64
> 
> I did have a lot of development packages installed, so the list may be incomplete for building.

Are these for running it, or just building it?

7eggert, 2021-12-01:
For running I need libQt5Charts5 only.

Also I forgot to say: I'm using OpenSuse Tumbleweed