marazmista/radeon-profile #263: After Setting ppfeaturemask to 0xffffffff I cannot set percentage overclock or set states overclock

Clueninja, 2021-04-09:
While running with ppfeaturemask=0xffffffff:
radeon-profile tries to apply states but in "general" no changes are observed.
Running overclock state commands in user terminal responds with "no permission".
After running sudo su, I can edit the state frequencies. However, I'd prefer not to run sudo su every boot up or create a script that has these privileges.
Changes made during su appear in radeon-profile "general"

However when running with default ppfeaturemask:
States changing is greyed out as expected but the percentage overclock is applied and changes appear in the "set manual frequencies section" I thought this was a bit odd.

Expected outcome while ppfeaturemask=0xffffffff is that I can change states and percentage overclock from the GUI.
Running AUR versions of radeon-profile and radeon-profile-daemon

Any help is greatly appreciated

Operating System: Manjaro Linux
KDE Plasma Version: 5.21.3
KDE Frameworks Version: 5.80.0
Qt Version: 5.15.2
Kernel Version: 5.9.16-1-MANJARO
OS Type: 64-bit
Graphics Platform: Wayland
Processors: 4 × Intel® Core™ i5-6500 CPU @ 3.20GHz
Memory: 7.7 GiB of RAM
Graphics Processor: Radeon RX550/550 Series
Graphics Driver amdgpu

https://unix.stackexchange.com/questions/404780/cant-overclock-gpu-using-amdgpu-driver-on-archlinux

[image]

Oxalin, 2023-01-27:
@Clueninja Is this still relevant? Why was it closed and then reopened?

Clueninja, 2023-01-27:
@Oxalin I accidently closed the issue, I use a different computer now so I havent tested it for a while

rizzini, 2023-07-21:
I get the same behavior.  Overclock values and states are not applied. RX 580 8GB here.