marazmista/radeon-profile #265: Fan profile not restored after sleep

yonatan8070, 2021-04-25:
Steps to reproduce:

1. Select "Custom curve" fan control
2. Put PC to sleep
3. Wake PC from sleep
4. "Custom curve" will still appear selected, but the fan curve will not actually control the fans on the graphics card (in my case, leading to over 60C while watching YouTube)

emerge-e-world, 2022-02-25:
I have applied a fix in my repo that should restore the fan profile after suspend. You may pull from https://github.com/emerge-e-world/radeon-profile and compile that version. There are also some other bug fixes in that repo.

Oxalin, 2023-01-27:
I merged the patches from emerge-e-world in my own fork and I'm building toward a new release from there. If you want to test Radeon Profile and Radeon Profile daemon (you need both up to date), have a go over here:

Radeon Profile: https://github.com/Oxalin/radeon-profile
Radeon Profile daemon: https://github.com/Oxalin/radeon-profile-daemon

@yonatan8070 , if this issue is fixed in either fork, could close the issue or at least let us know? Thank you