marazmista/radeon-profile #267: Missing graphics in buttons on left

arromdee, 2021-05-03:
I'm getting the same graphical problem as shown in #238 and #235 : the buttons on the left are things like "BOOTUP_DEFAULT and 3D_FULL_SCREEN rather than graphical icons.  Also, I can push them and they go down but they don't seem to do anything.

Built on a new installation of Mageia 8 64 bit, Radeon R7 260X.

Oxalin, 2023-01-27:
It is not an issue. Your card is using other power profiles. I can't remember when this was modified (which GPU generation), but it is as expected. Graphical icons could have been created to represent the new power profiles, but none were made. Instead, Radeon profile is using buttons with labels.

This could either be a feature request (replace by icons or use only labels) or it should be closed.

My1, 2023-07-01:
is it intended tho that the buttons dont really do anything? I try to enter POWER_SAVING to maybe get the GPU to run with less power but it immediately jumps back to 3D_FULL_SCREEN