marazmista/radeon-profile #268: Problem with multigpu setup

Norush89, 2021-05-10:
Hello

Everything was working gr8 for me until i installed another gpu trough riser card.
Now i am unable to control fan speeds and the program crashes occasionally.
i am a new linux user so any help with this issue would be appreciated.

Norush89, 2021-05-10:
as a quick fix, can i maybe open 2 instances of the program using a specific gpu only?

Oxalin, 2023-01-07:
Hi. Are you still using Radeon Profile with a dual GPU setup? Your bug report is "oldish", so I may be too late for you.

Radeon Profile doesn't properly support multi-GPU if I can say, because it will only apply the settings and profiles to a single GPU at a time. However, it should be possible to select which GPU you want to control and set it without any problem. That being said, you must be causious to not apply settings for a specific GPU on a different one when switching from GPU to the other.

I have a dual GPU setup, I've included some fixes in a new release I'm working on (https://github.com/Oxalin/radeon-profile). I have plans to better support multi-GPU setup, but I'm not there yet. You could try it and see if thing are better.