marazmista/radeon-profile #270: [BUG] State table clock ranges do not save with the profile

ryannathans, 2021-05-14:
When saving a state table profile the "set ranges" window does not save and must be manually set every time the profile is loaded.

For example, this radeon vii oc profile I have made has a peak clock of 1925mhz and such every time I apply the profile (almost every boot) I must manually open this window and change 1801 to 1925mhz and also bump up the max memory clock to 1200mhz.
[image]

Oxalin, 2023-01-27:
Is this still an issue? I could have a look at it under my fork.

ryannathans, 2023-01-27:
@Oxalin not sure, I swapped to using corectrl and a 6900XT and now I only have one voltage control lol