marazmista/radeon-profile #273: Segfault in dXorg::parseOcTable()

SytheZN, 2021-06-17:
Hey,

I'm running into a segfault on launch when parsing pp_od_clk_voltage (built off 0fd5b8dd326957e6fa8afda2bba5fc4c48fd0de4).
App works fine with amdgpu.ppfeaturemask unset, but crashes with it set to 0xffffffff.

OS: Fedora 34 Workstation
Kernel: 5.12.10-300.fc34.x86_64
GPU: XFX RX 6900XT

The specific crash location is:
[image]

```
$ cat /sys/class/drm/card0/device/pp_od_clk_voltage
OD_SCLK:
0: 500Mhz
1: 2529Mhz
OD_MCLK:
0: 97Mhz
1: 1000MHz
OD_VDDGFX_OFFSET:
0mV
OD_RANGE:
SCLK:     500Mhz       3000Mhz
MCLK:     674Mhz       1075Mhz
```

emerge-e-world, 2021-12-05:
This looks like the same issues I reported in #279. Can you try recompiling with this fix #280?

Oxalin, 2023-01-27:
Hi @SytheZN . Are you still experiencing the issue using either emerge-e-world's fix #280 or my own forks? I merged the patches from emerge-e-world and I'm building toward a new release from there.

Radeon Profile: https://github.com/Oxalin/radeon-profile
Radeon Profile daemon: https://github.com/Oxalin/radeon-profile-daemon

If this is fixed with either fork, could you close the issue?

SytheZN, 2023-01-31:
@Oxalin, thanks for the feedback, and my apologies for the delay.

I've moved off of radeon-profile onto other tools as I'd assumed this project was abandoned.
Happy to close off the issue, but I'm unable to confirm whether or not the underlying problem is solved.

closed 2023-01-31 by SytheZN