marazmista/radeon-profile #276: Still supported?

StatusCode404, 2021-09-10:
I can see that there's been no activity nor new tags nor the last tag put to stable since last year!

Is this still supported?

StatusCode404, 2021-09-15:
So no one knows?  It shows 5 contributors to this project?

StatusCode404, 2021-09-22:
@marazmista @Danysan1 
Just reaching out and wondering if you could answer the above?
Just wondering if there will be ongoing support for this project as newer radeon cards come out?

Danysan1, 2021-09-22:
Hi, it's a while that I have changed PC with another one with an Nvidia card, so unfortunately I am unable to develop and test changes on this project

StatusCode404, 2021-09-22:
> Hi, it's a while that I have changed PC with another one with an Nvidia card, so unfortunately I am unable to develop and test changes on this project

Thanks, do you know if anyone of your previous counterparts are still supporting this software?

emerge-e-world, 2022-02-25:
I am not one of the previous contributors, but I have started collection fixes from other pull requests and my own fixes and improvements in my fork, after I noticed that marazmista isn't currently active and no one is responding to PRs. I plan to keep working on it further. You find it at:
https://github.com/emerge-e-world/radeon-profile

StatusCode404, 2022-02-25:
> I am not one of the previous contributors, but I have started collection fixes from other pull requests and my own fixes and improvements in my fork, after I noticed that marazmista isn't currently [...]

Thanks for the heads up!
I noticed that your repo has no "issues" tab, how do users, contributors, testers; communicate within your 'active' repo?
Will master and tags be stable?  - I had a quick look and there are no new tags.

sr258, 2022-04-24:
Great that you've picked this up, @emerge-e-world ! It's a fantastic project that I really need to get my GPU fan to a bearable level. Do you plan on publishing Flatpaks and Snaps? At the moment I build everything myself (I need some bugfixes and am on Ubuntu 22.04, for which there isn't a PPA yet). It would be really great if there was an easy way to install Radeon Profile without adding a PPA!

emerge-e-world, 2022-05-18:
I haven't really thought about packaging yet. I wasn't planning on taking over the project at all, I merely hoped my changes would eventually be merged upstream. But it doesn't look like this is going to happen any time soon, so it would indeed be useful to have something easy to install. 
I'll have to look into snaps/flatpaks a bit, I don't have much experience with those. Creating a PPA should be easy enough though. I'd prefer native packages in distro repos overall. A lot of distros already have marazmista's version in their repos that worked well so far, it would be easy to build from my repo the same way.
Anyhow, I'll look into it more after I got some more fixes merged, it would certainly be nice to have a working version that is easy to install cross distro. For the moment you'll have to compile it yourself.

Oxalin, 2022-06-23:
I'm late to the game: I haven't had time to work or contribute more on radeon profile, but I'm still interested in it. I am one of the previous contributors and I'd like to contribute more. @marazmista could you let us know if you are still interested in this project. Could you have a (or a few) co-administrator? @emerge-e-world has a fork with some additional commits that could be brought back to the main project.

Oxalin, 2023-01-07:
I took sometime during the last two weeks to build a new version. I'm almost ready to release a new version. You can have a look here: https://github.com/Oxalin/radeon-profile

Oxalin, 2023-01-27:
If you want to test Radeon Profile and Radeon Profile daemon (you need both up to date), have a go over here:

Radeon Profile: https://github.com/Oxalin/radeon-profile
Radeon Profile daemon: https://github.com/Oxalin/radeon-profile-daemon

I'm almost ready for a release.

kolAflash, 2023-06-10:
@Oxalin  
Thanks!  
radeon-profile-20200824 segfaulted after upgrading my Notebook from Debian-11 to Debian-12.  
_(notebook hardware specs: HP EliteBook 735 G8, Ryzen 3500U incl. Radeon GPU)_
But [your version](https://github.com/Oxalin/radeon-profile/commit/32fd604ccba91bdb4f7c2e1fcd2b421e35ac214e) runs fine!  
Strangly it still runs fine with Debian-12 on my desktop computer.
_(desktop hardware specs: ASUS TUF GAMING B550-PLUS Mainboard, Ryzen 5800X CPU, Gigabyte Radeon RX 590 GPU)_

Would you mind opening the issue section in your projects?  
https://github.com/Oxalin/radeon-profile  
https://github.com/Oxalin/radeon-profile-daemon

And you also might want to search for these strings in your version:  
`marazmista/radeon-profile`  
At least add your GitHub URL too in the "about section" of the program.  
_(it's probably a good idea to keep the original author for reference and copyright reasons)_