marazmista/radeon-profile #278: Translation not found. QFSFileEngine::open: No file name specified

kadirilgin1453, 2021-11-21:
I use arch linux. sudo radeon profile connmand and 
Translation not found.
QFSFileEngine::open: No file name specified

Error please help me

Shadowolf7, 2021-11-29:
I also see the same error message when attempting to switch to card1 which is my rx560x.

sidenote; the application requires elevated privileges and thus I must start it from bash with gnomesu,

UltimateByte, 2021-12-01:
Hello,
Kind of the same issue here.

Arch Linux with KDE Plasma.
```
[lr@arch log]$ uname -a
Linux arch 5.15.5-arch1-1 #1 SMP PREEMPT Thu, 25 Nov 2021 22:09:33 +0000 x86_64 GNU/Linux
```
Program version:
```
[lr@arch ~]$ yay -Q | grep radeon-profile
radeon-profile-git 20200824.r6.g0fd5b8d-1
```

Running as the user, radeon-profile starts normally but shows a blank program.
[image]

Running as root `sudo radeon-profile` did show relevant information but overclocking didn't work. Though, in my case, I would like to downclock the GPU, and slider doesn't seem to allow that; furthermore, I was unable to edit clock values manually.

Since overclocking didn't seem to work, I've edited /etc/default/grub as per [Arch's Wiki](https://wiki.archlinux.org/title/AMDGPU#Boot_parameter):
`GRUB_CMDLINE_LINUX_DEFAULT="loglevel=3 quiet amdgpu.ppfeaturemask=0xffffffff"`

After rebooting, parameter seems to be taken into account:
```
[lr@arch ~]$ printf "0x%08x\n" $(cat /sys/module/amdgpu/parameters/ppfeaturemask)
0xffffffff
```

But I am now unable to restart the program as root:
```
[lr@arch ~]$ sudo radeon-profile
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
QFSFileEngine::open: No file name specified
Segmentation fault
```

Got a log from Xorg.0.log, but seems irrelevant:
```
[   851.783] (--) AMDGPU(0): HDMI max TMDS frequency 600000KHz
[   851.784] (II) AMDGPU(0): EDID vendor "ACI", prod id 10151
[   851.784] (II) AMDGPU(0): Using hsync ranges from config file
[   851.784] (II) AMDGPU(0): Using vrefresh ranges from config file
[   851.784] (II) AMDGPU(0): Printing DDC gathered Modelines:
[   851.784] (II) AMDGPU(0): Modeline "2560x1440"x0.0  586.00  2560 2568 2600 2640  1440 1465 1473 1543 +hsync +vsync (222.0 kHz eP)
[   851.784] (II) AMDGPU(0): Modeline "2560x1440"x0.0  241.50  2560 2608 2640 2720  1440 1443 1448 1481 +hsync +vsync (88.8 kHz e)
[   851.784] (II) AMDGPU(0): Modeline "1280x720"x0.0   74.25  1280 1390 1430 1650  720 725 730 750 +hsync +vsync (45.0 kHz e)
[   851.784] (II) AMDGPU(0): Modeline "1280x720"x0.0   74.25  1280 1720 1760 1980  720 725 730 750 +hsync +vsync (37.5 kHz e)
[   851.784] (II) AMDGPU(0): Modeline "2560x1440"x0.0  482.63  2560 2568 2600 2640  1440 1447 1455 1525 +hsync +vsync (182.8 kHz e)
[   851.784] (II) AMDGPU(0): Modeline "720x576"x0.0   27.00  720 732 796 864  576 581 586 625 -hsync -vsync (31.2 kHz e)
[   851.784] (II) AMDGPU(0): Modeline "1920x1080"x0.0  148.50  1920 2448 2492 2640  1080 1084 1089 1125 +hsync +vsync (56.2 kHz e)
[   851.784] (II) AMDGPU(0): Modeline "1920x1080"x0.0  148.50  1920 2008 2052 2200  1080 1084 1089 1125 +hsync +vsync (67.5 kHz e)
[   851.784] (II) AMDGPU(0): Modeline "1440x480"x0.0   54.00  1440 1472 1596 1716  480 489 495 525 -hsync -vsync (31.5 kHz e)
[   851.784] (II) AMDGPU(0): Modeline "1440x576"x0.0   54.00  1440 1464 1592 1728  576 581 586 625 -hsync -vsync (31.2 kHz e)
[   851.784] (II) AMDGPU(0): Modeline "800x600"x0.0   40.00  800 840 968 1056  600 601 605 628 +hsync +vsync (37.9 kHz e)
[   851.784] (II) AMDGPU(0): Modeline "800x600"x0.0   36.00  800 824 896 1024  600 601 603 625 +hsync +vsync (35.2 kHz e)
[   851.784] (II) AMDGPU(0): Modeline "640x480"x0.0   31.50  640 656 720 840  480 481 484 500 -hsync -vsync (37.5 kHz e)
[   851.784] (II) AMDGPU(0): Modeline "640x480"x0.0   30.24  640 704 768 864  480 483 486 525 -hsync -vsync (35.0 kHz e)
[   851.784] (II) AMDGPU(0): Modeline "640x480"x0.0   25.18  640 656 752 800  480 490 492 525 -hsync -vsync (31.5 kHz e)
[   851.784] (II) AMDGPU(0): Modeline "720x400"x0.0   28.32  720 738 846 900  400 412 414 449 -hsync +vsync (31.5 kHz e)
[   851.784] (II) AMDGPU(0): Modeline "1280x1024"x0.0  135.00  1280 1296 1440 1688  1024 1025 1028 1066 +hsync +vsync (80.0 kHz e)
[   851.784] (II) AMDGPU(0): Modeline "1024x768"x0.0   78.75  1024 1040 1136 1312  768 769 772 800 +hsync +vsync (60.0 kHz e)
[   851.784] (II) AMDGPU(0): Modeline "1024x768"x0.0   75.00  1024 1048 1184 1328  768 771 777 806 -hsync -vsync (56.5 kHz e)
[   851.784] (II) AMDGPU(0): Modeline "1024x768"x0.0   65.00  1024 1048 1184 1344  768 771 777 806 -hsync -vsync (48.4 kHz e)
[   851.784] (II) AMDGPU(0): Modeline "832x624"x0.0   57.28  832 864 928 1152  624 625 628 667 -hsync -vsync (49.7 kHz e)
[   851.784] (II) AMDGPU(0): Modeline "800x600"x0.0   49.50  800 816 896 1056  600 601 604 625 +hsync +vsync (46.9 kHz e)
[   851.784] (II) AMDGPU(0): Modeline "800x600"x0.0   50.00  800 856 976 1040  600 637 643 666 +hsync +vsync (48.1 kHz e)
[   851.784] (II) AMDGPU(0): Modeline "1440x900"x0.0  106.50  1440 1520 1672 1904  900 903 909 934 -hsync +vsync (55.9 kHz e)
[   851.785] (II) AMDGPU(0): Modeline "1280x1024"x0.0  108.00  1280 1328 1440 1688  1024 1025 1028 1066 +hsync +vsync (64.0 kHz e)
[   851.785] (II) AMDGPU(0): Modeline "1152x864"x0.0  108.00  1152 1216 1344 1600  864 865 868 900 +hsync +vsync (67.5 kHz e)
```

Hope that helps sorting the issue out. I'll do my best to stay available and help as I can. 
Though, please note that I'm coming from Fedora as a daily driver Linux desktop on a computer with no dedicated graphis card, as my background is to work as a sysadmin on CentOS and Debian servers. So even though I've got some specific Linux knowledge, I'm very new to Arch as well as dealing with AMD GPUs on Linux. Therefore, details might be useful for any testing or diagnostics request.

Thanks for reading, and best of luck to anyone trying to fix this as well.

fischer-felix, 2021-12-18:
Any developments on this?
I have the same issue on my system (also Arch Linux, stable branches), but it works fine in Arch in VirtualBox (relatively clean install, just for testing).

fischer-felix, 2021-12-18:
Update: The program starts, as long as you do so without the systemd daemon running or without sudo, so it is not a problem with the actual frontend I guess.

eitch, 2022-01-16:
I've also got this issue on Pop OS 21.10. I compiled from source, even trying https://github.com/emerge-e-world/radeon-profile getting the same error

fischer-felix, 2022-01-16:
I think QFSFileEngine::open: No file name specified is not actually the cause of it not launching, since it is working for me, but still prints it. 
It seems to be something about loading the last set config or something (just guessing).

Natim, 2022-01-26:
```
Thread 1 "radeon-profile" received signal SIGSEGV, Segmentation fault.
0x00007ffff6947b30 in QString::toUInt(bool*, int) const () from /lib/x86_64-linux-gnu/libQt5Core.so.5
```

phreeky82, 2022-03-05:
> please try the current version from by fork at https://github.com/emerge-e-world/radeon-profile again

Thanks this works! At least basic functionality (fan control) tested OK.

emerge-e-world, 2022-03-06:
Glad to hear, thanks for testing! What card do you have?
Also could you kindly post your pp_od_clk_voltage tables? This could help me figuring out what may be missing for the overclocking bits. Just can get the contents with:
`cat /sys/class/drm/card0/device/pp_od_clk_voltage`

eitch, 2022-03-30:
@emerge-e-world A bit late, but here:

    $ cat /sys/class/drm/card0/device/pp_od_clk_voltage
    OD_SCLK:
    0: 500Mhz
    1: 2374Mhz
    OD_MCLK:
    0: 97Mhz
    1: 1000MHz
    OD_VDDGFX_OFFSET:
    0mV
    OD_RANGE:
    SCLK:     500Mhz       2800Mhz
    MCLK:     674Mhz       1075Mhz

Endoril, 2022-08-28:
Thank you @emerge-e-world , i have the same problem and your fork is working !
Do you plan to merge your modifications in radeon-profile ?

emerge-e-world, 2022-09-07:
Glad it's working and thanks for the table. I made pull requests for upstream radeon-profile, but marazmista  seems to be inactive for quite a while now. For the moment I plan to continue working on my fork, collect bug fixes there and sort out some more Navi and UI issues. But I'd be glad if my fixes would be merged eventually.

Oxalin, 2023-01-27:
I merged the patches from emerge-e-world in my own fork and I'm building toward a new release from there. If you want to test Radeon Profile and Radeon Profile daemon (you need both up to date), have a go over here:

Radeon Profile: https://github.com/Oxalin/radeon-profile
Radeon Profile daemon: https://github.com/Oxalin/radeon-profile-daemon