marazmista/radeon-profile #286: segfault when trying to run radeon-profile on Radeon Vega 8

Koi-foo, 2022-09-13:
Hello.
On laptop:
Acer Aspire 3 A315-42-R94P
CPU: AMD Ryzen 5 3500U 2100
GPU: AMD Radeon Vega 8

Shell: KDE 5.24.6 \ X11 \ Qt:5.15.4
Kernel: 5.10.*

When I try to run radeon-profile, I get a segmentation error message.
dmesg shows:
[ 3372.868718] amdgpu: pp_dpm_get_sclk_od was not implemented.
[ 3372.868774] amdgpu: pp_dpm_get_mclk_od was not implemented.
[ 3372.877299] traps: radeon-profile[7131] general protection fault ip:7fec96bececc sp:7ffe9350f450 error:0 in libQt5Core.so.5.15.2[7fec96b0d000+31c000]

Can you tell me if this can be fixed?

MasterCATZ, 2022-09-27:
similar thing for me 

worked for my "RX 570 + Vega 64" last year 
and Radeon-profile was working just fine while a was using a R9 290
and just swapped in a 6800 XT then radeon-profile just stopped working with Segmentation fault's 

tho mine say 

~~~
[43655.382602] traps: radeon-profile[2950791] general protection fault ip:7fa5564dee40 sp:7ffdf871f910 error:0 in libQt5Core.so.5.15.3[7fa556416000+30f000]
[43683.620061] traps: radeon-profile[2951959] general protection fault ip:7f3679d9ee40 sp:7ffd0d63fc20 error:0 in libQt5Core.so.5.15.3[7f3679cd6000+30f000]

~~~

currently switched to using https://gitlab.com/corectrl/corectrl
as it seem the maintainer here has not done any code changes for 2 years now  but I do prefer this program still

Koi-foo, 2022-11-03:
> аналогичная вещь для меня
> 
> работал на моем «RX 570 + Vega 64» в прошлом году , и профиль Radeon работал нормально, пока я использовал R9 290 и просто заменил его на 6800 XT, затем профиль radeon просто перестал работать с ошибкой сегментации.
> 
> мой говорит
> 
> ```
> [43655.382602] traps: radeon-profile[2950791] general protection fault ip:7fa5564dee40 sp:7ffdf871f910 error:0 in libQt5Core.so.5.15.3[7fa556416000+30f000]
> [43683.620061] traps: radeon-profile[2951959] general protection fault ip:7f3679d9ee40 sp:7ffd0d63fc20 error:0 in libQt5Core.so.5.15.3[7f3679cd6000+30f000]
> ```
> 
> в настоящее время перешел на использование https://gitlab.com/corectrl/corectrl , так как кажется, что сопровождающий здесь не вносил никаких изменений в код в течение 2 лет, но я все еще предпочитаю эту программу
> 
> попробуй форк отсюда
> 
> [https://github.com/emerge-e-world/radeon-профиль](https://github.com/emerge-e-world/radeon-profile)

Thanks a lot. He works well.

Oxalin, 2023-01-27:
emerge-e-world patches were included in my fork and I added new features and fixes along the way. I'm preparing a new release.

If you want to test my fork: https://github.com/Oxalin/radeon-profile
Make sure you also build and run the latest radeon-profile-daemon: https://github.com/Oxalin/radeon-profile-daemon