marazmista/radeon-profile #290: fan profile resets to default as soon as the window is closed

FantasyPvP, 2023-01-02:
as soon as i close the radeon profile window (where my fans are set at 10% below a certain temperature) the fans turn off (im assuming reset to their default profile)

the custom fan curve returns as soon as i open the window again. I have the daemon running in the background which has been enabled using systemctl. Can i make it so that the fan curve persists even when i close the app?

Oxalin, 2023-01-07:
Hey. Can you tell me more about your configuration? Are you using the daemon?

FantasyPvP, 2023-01-16:
> Hey. Can you tell me more about your configuration? Are you using the daemon?

yes, i have the deamon enabled as a service, the app doesnt seem to function without it, i installed both of them through the AUR

Oxalin, 2023-01-17:
Sorry, I had skipped the running daemon info.

I'll look into it in the next few days. I'll keep in touch.

Oxalin, 2023-01-27:
What is your card?

Also, I'm testing with a Vega FE and it seems to follow properly my profile when RP is running, but not visible (closing the window make it hidden, but the program should still be running). Are you closing the app completely (quit) or is the client still running (app icon may be visible in your taskbar). If the client is closed (quit), the GPU is returning to its default behavior and this is as intended. If I remember correctly from what I read somewhere (I can't remember if this was from an older issue, in the code or in the README), it is mostly to prevent unexpected behavior when unattended if the fan profile is too low for a given temp or if an overclock profile was to cause problem. As long as the RP is not running, no profile are applied and the card is managing itself.

FantasyPvP, 2023-03-12:
> What is your card?
>
> Also, I'm testing with a Vega FE and it seems to follow properly my profile when RP is running, but not visible (closing the window make it hidden, but the program should still [...]

im running a 6700xt (sorry for not replying sooner, i look at my github notifications a lot)

closed 2023-03-12 by FantasyPvP