marazmista/radeon-profile #298: Window appears, but can't select any GPU

RafaelLinux, 2023-10-19:
I installed "radeon-profile" (no the daemon) but when I launch it, has not GPUs availables to select and any meter is showed. I'm using RX 

[image]

"radeontop" shows meters, but "UNKNOW" chip!!!
[image]

"nvtop" shows correctly GPU chip and graphics (but not so complete as "radeon-profile".
[image]

So, why "radeon-profile" not monitoring anything?

Using openSUSE Thumbleweed

Thank you

t0m5k1, 2023-10-23:
To use it I have to run it as root, i made a change to the .desktop file to account for it and all is well

RafaelLinux, 2023-10-25:
For reasons I don't know, launching it as root from CLI, it shows errors:

```
>sudo radeon-profile
[sudo] contraseña para root: 
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, xcb.

```

However, if I launch it as root with the "kdesu" command (equivalent to launch an application with `root` privileges in Plasma), it works fine.  Then, I noticed two things:

- No graphs
[image]
- Text using "monospace serif" like in "General" tab (seems "courier" font) is slightly less readable in my HiDPI screen. Could you increase its font size 1 or 2 points?

Thanks

loligator, 2024-01-26:
Installed with dnf on Fedora and ran into the same issue. It looks like there are a lot of negative reviews on the software center as well... maybe should change the .desktop file that installs by default or else few are going to have any success.